UOsaka-Harada-Laboratory/onrobot #1: compatibility with onrobot gripper 2FG7

imlaras, 2022-04-22:
Hello, this is not an issue but a question. Would this software be compatible with the onrobot gripper model 2FG7?

takuya-ki, 2022-04-22:
@imlaras Thank you for your contact!
To be honest, I am not sure because the model was not tested with this.
I recommend that you first check if you can simply control the gripper via an onrobot compute box with Modbus TCP while referring to another repository [onrobot-rg](https://github.com/takuya-ki/onrobot-rg).

imlaras, 2022-04-27:
Hi @takuya-ki thank you so much for your earlier responde! I do not have an onrobot compute box to check it out. Have you been capable of controlling it directly through the robot's TCP connector? Or do you always (with and without ros) need a compute box to communicate to the gripper?

takuya-ki, 2022-05-01:
@imlaras Sorry to confuse you. The below repositories were developed for the gripper with a tool changer connecting to a compute box with an ethernet cable.
In README, I have added the description of the hardware configuration that I assume.

closed 2022-05-09 by imlaras