UOsaka-Harada-Laboratory/onrobot #5: AttributeError: 'ModbusIOException' object has no attribute 'registers'

Robertomendivil97, 2022-11-09:
Hi, the problem is on my TCP Connection

Error:
AttributeError: 'ModbusIOException' object has no attribute 'registers'
After trying to run either roslaunch 

Python 3.8,10
pymodbus==2.5.3

The computer box is working because I can send data through the web client, so IP are configured correctly
Do you know the solution?

Robertomendivil97, 2022-11-10:
Because I am using the Dual quicker changer, the register is 66 (and 67 for second gripper) instead of 65
Changing that it works
[image]

Awesome job with this repo!

takuya-ki, 2022-11-11:
@Robertomendivil97 
Thank you for the investigation! 
I updated the [master branch](https://github.com/Osaka-University-Harada-Laboratory/onrobot/tree/db60577168ed6db8963b76b48e134aac175d1bfc) with launch files and nodes that uses a rosparam named changer_addr.
If you are interested, please try to use it with your dual quick changer.
Please let me know if you have any other suggestions!

Robertomendivil97, 2022-11-11:
Thank you very much for the really quick update!
The parameters works and I can chose which tool.
Some of the suggestions are:
- With the current update I can control one of the grippers but not at the same time, probably other node can be implemented for that.
--A nice really feature to have would be the restart power cycle, because if the contact buttons on the end grippers are detected, you need to restart the power cycle from the web client. A ros service could be an option for that.

Excellent job with this ROS package

takuya-ki, 2022-11-11:
@Robertomendivil97 
Thank you for the suggestion! The problem is that I do not have the dual quick changer now. 
If possible, could you please update and test it so and then pull request?

Robertomendivil97, 2022-11-15:
Of course
This week will continue with the testing and could pull request

BrettRD, 2023-09-01:
There's a type error here that could be more informative
(I'm working on a ROS2 port for the RG6 on a UR5e and I'm getting a lot of timeouts that trigger this error)

The developer of pymodbus decided that read_holding_registers should return the exception instead of throwing it
https://github.com/pymodbus-dev/pymodbus/issues/1464
So you can't do a try catch, you have to test the return type
https://github.com/DavidYaonanZhu/robotiq/commit/fbcd1052314574da0e7ed32548db510d84edc42d