UOsaka-Harada-Laboratory/onrobot #9: No response from the OnRobot Gripper

ericahellscythe, 2023-02-13:
Hi,

I tried to rosrun the controller to control the rg2 but whenever i try to run a command, nothing works. There is no error. It just fails to execute. Any ideas why?

[image]

takuya-ki, 2023-02-13:
@ericahellscythe Thanks for the question.
Let me clarify the hardware configuration. Does the wiring meet the following specifications?

```
1. Connect the cable between Compute Box and Tool Changer
2. Connect an ethernet cable between Compute Box and your computer
```

Also, if you have time, please try to control the gripper with below repository without ROS.
https://github.com/takuya-ki/onrobot-rg

ericahellscythe, 2023-02-13:
Hi! Thanks for the quick reply.
I connected the grippers to the compute box and the compute box to the robot and my pc.

[image]

I tried running your repository but got this error.

[image]

takuya-ki, 2023-02-13:
@ericahellscythe 
Please be sure to install pymodbus==2.5.3 and execute the script with python3.
Also, just confirm whether the red lighting of quick changer's led is normal. You might need to replug it.

ericahellscythe, 2023-02-13:
Realized another issue had the same error as me. I managed to resolve it. It opens and closes like it's suppose to. Also managed to resolve the red light back to normal. How do I proceed from here now?

takuya-ki, 2023-02-13:
@ericahellscythe Good!
Then, please try the execution of controller with ROS again. Maybe, the error was occuring due to the same issue.

ericahellscythe, 2023-02-13:
Hi, my roslaunch for the bringup can't run right now. I now have the ModbusIOException object has no attribute registers error. I tried to change the register as mentioned in the other issue but this time it did not work. Any ideas?

takuya-ki, 2023-02-13:
@ericahellscythe 
Are you working on it with ROS Noetic? Did you correctly install all dependencies with below commands explained in README?
```
cd catkin_ws/src
git clone https://github.com/takuya-ki/onrobot.git --depth 1
git clone https://github.com/roboticsgroup/roboticsgroup_upatras_gazebo_plugins.git --depth 1
cd ../
sudo rosdep install --from-paths ./src --ignore-packages-from-source --rosdistro noetic -y --os=ubuntu:focal -y
sudo apt install ros-noetic-ros-control ros-noetic-ros-controllers
catkin build -DPYTHON_EXECUTABLE=/usr/bin/python3
```

ericahellscythe, 2023-02-13:
Nvm I didnt save the changing of the register address haha. It works! Thanks alot. 

By the way is it possible to change the force of the grip? I see that the command allows me to open (o), close (c) or even have a specific width i want. But can i also control the force?

takuya-ki, 2023-02-13:
@ericahellscythe Ok. Please carefully read the programs and find your mistakes before asking here.
It is possible to control the force.
https://github.com/Osaka-University-Harada-Laboratory/onrobot/blob/main/onrobot_rg_control/msg/OnRobotRGOutput.msg

closed 2023-02-13 by takuya-ki