UOsaka-Harada-Laboratory/onrobot #13: How to setup modbus on TP for RG2

masterdeki, 2023-10-09:
Hi
I'm trying to control RG2 gripper but I need help in setting up modbus connection?
Is there some manual or tutorial on how to properly setup modbus on teach pendant?

takuya-ki, 2023-10-09:
@masterdeki 
Are you talking about Universal Robots' teach pendant? Please check the README of this repository carefully.
Since this package assumes a configuration of your computer - ethernet cable - compute box - tool changer - gripper, the teach pendant is not included in the expected hardware configuration.

masterdeki, 2023-10-09:
Yes, talking about ur5e TP

takuya-ki, 2023-10-09:
@masterdeki 
If that is the case, sorry but this issue is not related to this repository so I close this.

closed 2023-10-09 by takuya-ki