UOsaka-Harada-Laboratory/onrobot #14: Control RG6 Gripper mounted on UR5 robot arm

binhkhongbeo, 2023-10-10:
Hello there,
I would like to control the OnRobot RG6 gripper that mounted to a UR5 robot arm via ROS noetic. But now I am having a problem in connecting my PC to robot through Ethernet.

UR5 was running just fine based on universal_robot's instruction (without RG6) in the link: https://github.com/UniversalRobots/Universal_Robots_ROS_Driver
Then I tried to control RG6 based on instruction in the link: https://github.com/Osaka-University-Harada-Laboratory/onrobot but still use the Ethernet Connection & External Control script(URCap) like UR5 case.
Firstly, I assumed the gripper and the arm share the shame IP so I use the robot IP, and it ran into error.
I keep trying by changing the robot IP to the same IP as default IP of RG6 Gripper, which is 192.168.1.1 and default port: 502, but nothing cool happen and eventually I got this ERROR which i could not fix :(

ERROR:
AttributeError: 'ExceptionResponse' object has no attribute 'registers'
[DEBUG] [1696923112.318935]: [/OnRobotRGInput] failed to receive incoming message : unable to receive data from sender, check sender's logs for details
[OnRobotRGTcpNode-2] process has died [pid 14410, exit code 1, cmd /home/mngu/catkin_ws/devel/lib/onrobot_rg_control/OnRobotRGTcpNode.py __name:=OnRobotRGTcpNode __log:=/home/mngu/.ros/log/a4614100-673e-11ee-9256-03bb5f0dd7a5/OnRobotRGTcpNode-2.log].
log file: /home/mngu/.ros/log/a4614100-673e-11ee-9256-03bb5f0dd7a5/OnRobotRGTcpNode-2*.log
[OnRobotRGTcpNode-2] restarting process

takuya-ki, 2023-10-16:
@binhkhongbeo 
Since this package assumes a configuration of your computer - ethernet cable - compute box - tool changer - gripper, any UR-series arms are not included in the expected hardware configuration. Sorry the issue you're facing is not related to this repository.

closed 2023-10-16 by takuya-ki

Leaderleade, 2025-03-28:
Hello there, is there any possible solution to control RG6 on UR5 via this set up? Or do you know any repo that may help on this configuration?

Sett up: RG6 directly mount and connect to UR5 -> compute box -> ethernet cable -> PC, I do not have a tool changer unfortunately. I connect the cable of gripper directly on the arm.

Thanks in advance.